qualcomm/fastrpc #100: fastrpc: please rework the DSP directories location.

lumag, 2024-11-03:
In order to migrate from mounting dspso.bin at a fixed location to a properly installed binaries (https://github.com/linux-msm/dsp-binaries ) please extend FastRPC daemon / libraries to support specifying the basedir for the DSP binaries. The `/dsp` and `/vendor` locations must be ignored on a Linux-based installation. If no basedir was specified at the runtime, the default should be `/usr/share/qcom/dsp`.

quic-ekangupt, 2025-04-22:
Planning to handle the dsp-binaries path from service files. Will be sharing the changes soon.

lumag, 2025-05-24:
Note, this applies to userspace libraries too. Otherwise programs will still look for DSP libraries under `/vendor` or `/dsp`

quic-vkatoch, 2025-12-23:
Resolved via PR #234

closed 2025-12-23 by quic-vkatoch