qualcomm/fastrpc #140: FARF DSP logs aren't working

anonymix007, 2025-02-16:
I'm trying to test logging using a slightly modified calculator example (namely, `#define FARF_HIGH 1` just before the `#include "HAP_farf.h"`), however, can't see relevant messages in the `journalctl` logs:
```
calculator[46175]: fastrpc_context.c:299: fastrpc_context_table_init done
calculator[46175]: fastrpc_config.c:380: Reading configuration file: calculator.debugconfig
calculator[46175]: fastrpc_config.c:418: fastrpc_config_init: Couldn't find file calculator.debugconfig, errno (No such file or directory) at ., /usr/lib/dsp//cdsp, /usr/lib/dsp/,
calculator[46175]: fastrpc_apps_user.c:4256: fastrpc_apps_user_init done. default domain:3 and &fastrpc_trace:0xffff8b8d07a0
calculator[46175]: fastrpc_apps_user.c:4377: multidsplib_env_init: libcdsprpc.so loaded
calculator[46175]: fastrpc_apps_user.c:2703: remote_session_control Unsigned PD enable 1 request for domain 3
calculator[46175]: fastrpc_apps_user.c:2858: remote_session_control Register PD status notification request for domain 3
calculator[46175]: fastrpc_apps_user.c:3533: Successfully opened /usr/lib/dsp/cdsp/fastrpc_shell_unsigned_3, domain 3
calculator[46175]: fastrpc_apps_user.c:3641: Created user PD on domain 3, dbg_trace 0x0, enabled attr=> RPC timeout:0, Dbg Mode:N, CRC:N, Unsigned:Y, Signed:N, Adapt QOS:N, PD dump: (Config:N, Dbg:N), Perf: (Kernel:N, DSP:N), Iregion:N, QTF:N, UAF:N userPD initmem len:0x500000, Log pkt: N
calculator[46175]: fastrpc_apps_user.c:880: Warning 0xffffffff: get_lib_name failed for uri (null)
calculator[46175]: fastrpc_apps_user.c:3593: Error 0x14: fastrpc_enable_kernel_optimizations failed for domain 3 (Success)
calculator[46175]: fastrpc_perf.c:323: fastrpc_perf_init: enabled systrace 0x0 and RPC traces (kernel 0, dsp 0) with frequency 1000
calculator[46175]: fastrpc_apps_user.c:880: Warning 0xffffffff: get_lib_name failed for uri (null)
calculator[46175]: log_config.c:396: file_watcher_thread starting for domain 3
calculator[46175]: fastrpc_apps_user.c:880: Warning 0xffffffff: get_lib_name failed for uri (null)
calculator[46175]: fastrpc_apps_user.c:880: Warning 0xffffffff: get_lib_name failed for uri (null)
calculator[46175]: fastrpc_apps_user.c:1535: Error 0x8000043e: remote_handle64_invoke failed for module (null), handle 0xffff7c000b70, method 5 on domain 3 (sc 0x5000000) (errno Success)
calculator[46175]: listener_android.c:351: Error 0x8000043e: listener_start_thread domains support not available in listener
calculator[46175]: fastrpc_apps_user.c:1535: Error 0x8000043e: remote_handle64_invoke failed for module (null), handle 0xaaaae5558c50, method 2 on domain 3 (sc 0x2020200) (errno Success)
calculator[46175]: fastrpc_apps_user.c:1710: Warning 0x8000043e: remote_handle_open_domain: remotectl1 domains not supported for domain 3
calculator[46175]: listener_android.c:131: listener thread starting
calculator[46175]: fastrpc_apps_user.c:880: Warning 0xffffffff: get_lib_name failed for uri (null)
calculator[46175]: fastrpc_apps_user.c:880: Warning 0xffffffff: get_lib_name failed for uri (null)
calculator[46175]: fastrpc_mem.c:711: Error 0xe: remote_mmap64_internal failed for fd 0xc of size 262144 (flags 0x0, vaddrin 0xffff8a580000) errno Invalid argument
calculator[46175]: apps_mem_imp.c:182::error: 14: AEE_SUCCESS == (nErr = remote_mmap64_internal(fd, rflags, (uint64_t)buf, len, (uint64_t *)vadsp))
calculator[46175]: apps_mem_imp.c:206:Error 0xe: apps_mem_request_mmap64 failed for fd 0xc of size 262144 (lflags 0x80000000, rflags 0x0)
calculator[46175]: mod_table.c:909: Error 0xe: open_mod_table_handle_invoke failed for handle:0x8b8d6c30, sc:0x2010100
calculator[46175]: apps_std_imp.c:1035: Entering apps_std_fopen_with_env
calculator[46175]: apps_std_imp.c:1111: Successfully opened file libcalculator_skel.so
calculator[46175]: fastrpc_apps_user.c:1535: Error 0x8000043f: remote_handle64_invoke failed for module (null), handle 0xffff84000ca0, method 4 on domain 3 (sc 0x4010100) (errno Success)
calculator[46175]: adspmsgd.c:119::error: -2147482561: AEE_SUCCESS == (nErr = adspmsgd_adsp1_init3(handle, 0, RPCMEM_HEAP_DEFAULT, filter, DEFAULT_MEMORY_SIZE, &vapps))
calculator[46175]: adspmsgd.c:145:Error 0x8000043f: adspmsgd_init failed for handle 0xffff84000ca0 filter 2031647 with errno Success
calculator[46175]: log_config.c:219:adspmsgd not supported. nErr=8000043f
calculator[46175]: log_config.c:223:Found calculator.farf. adspmsgd enabled
calculator[46175]: fastrpc_apps_user.c:1836: remote_handle64_open: Successfully opened handle 0xaaaae5558d70 (remote 0x1eaa9f0) for file:///libcalculator_skel.so?calculator_skel_handle_invoke&_modver=1.0&_idlver=1.2.3&_dom=cdsp on domain 3 (spawn time 19965 us, load time 9666 us), num handles 1
calculator[46175]: fastrpc_apps_user.c:1836: remote_handle64_open: Successfully opened handle 0xaaaae5558e90 (remote 0x1eaae30) for file:///libcalculator_skel.so?calculator_skel_handle_invoke&_modver=1.0&_idlver=1.2.3&_dom=cdsp on domain 3 (spawn time 1 us, load time 1043 us), num handles 2
calculator[46175]: fastrpc_apps_user.c:1889: remote_handle_close_domain: closed module with handle 0x1eaa9f0 (skel unload time 1059 us)
calculator[46175]: fastrpc_apps_user.c:841: List of open handles on domain 3:
calculator[46175]: fastrpc_apps_user.c:846: libcalculator_skel.so, handle 0x1eaae30
calculator[46175]: fastrpc_apps_user.c:846: libcalculator_skel.so, handle 0x1eaa9f0
calculator[46175]: fastrpc_apps_user.c:1959: remote_handle64_close: closed module libcalculator_skel.so with handle 0xaaaae5558d70 remote handle 0x1eaa9f0, num of open handles: 1
calculator[46175]: fastrpc_apps_user.c:1959: remote_handle64_close: closed module libcalculator_skel.so with handle 0xaaaae5558e90 remote handle 0x1eaae30, num of open handles: 0
calculator[46175]: fastrpc_apps_user.c:3176: domain_deinit for domain 3: dev 6
calculator[46175]: listener_android.c:306: Error 0x8000040d: listener response with result 0x0 for ctx 0x739a0e, handle 0x8b8d6d30, sc 0xffffffff failed : listener thread exited (errno Success)
calculator[46175]: fastrpc_notif.c:104: Warning: fastrpc_notif_domain_deinit: unable to find hash-node for domain 3
calculator[46175]: log_config.c:440:Warning: file_watcher_thread received exit for domain 3, file calculator.farf
calculator[46175]: log_config.c:513: file_watcher_thread exiting for domain 3
calculator[46175]: fastrpc_apps_user.c:3388: close_device_node: closed dev 6 on domain 3
calculator[46175]: fastrpc_apps_user.c:3225: domain_deinit: closed device 6 on domain 3 (kill time 2082 us)
calculator[46175]: fastrpc_apps_user.c:3890: close_dev: unloading library libcdsprpc.so
calculator[46175]: fastrpc_apps_user.c:3176: domain_deinit for domain 0: dev -1
calculator[46175]: fastrpc_apps_user.c:3176: domain_deinit for domain 1: dev -1
calculator[46175]: fastrpc_apps_user.c:3176: domain_deinit for domain 2: dev -1
calculator[46175]: fastrpc_apps_user.c:3176: domain_deinit for domain 3: dev -1
calculator[46175]: fastrpc_apps_user.c:3176: domain_deinit for domain 4: dev -1
calculator[46175]: fastrpc_apps_user.c:3176: domain_deinit for domain 5: dev -1
calculator[46175]: fastrpc_apps_user.c:3176: domain_deinit for domain 6: dev -1
calculator[46175]: fastrpc_apps_user.c:3176: domain_deinit for domain 7: dev -1
calculator[46175]: fastrpc_apps_user.c:3176: domain_deinit for domain 8: dev -1
calculator[46175]: fastrpc_apps_user.c:3176: domain_deinit for domain 9: dev -1
calculator[46175]: fastrpc_apps_user.c:3176: domain_deinit for domain 10: dev -1
calculator[46175]: fastrpc_apps_user.c:3176: domain_deinit for domain 11: dev -1
calculator[46175]: fastrpc_apps_user.c:3176: domain_deinit for domain 12: dev -1
calculator[46175]: fastrpc_apps_user.c:3176: domain_deinit for domain 13: dev -1
calculator[46175]: fastrpc_apps_user.c:3176: domain_deinit for domain 14: dev -1
calculator[46175]: fastrpc_apps_user.c:3176: domain_deinit for domain 15: dev -1
calculator[46175]: fastrpc_apps_user.c:3176: domain_deinit for domain 16: dev -1
calculator[46175]: fastrpc_apps_user.c:3176: domain_deinit for domain 17: dev -1
calculator[46175]: fastrpc_apps_user.c:3176: domain_deinit for domain 18: dev -1
calculator[46175]: fastrpc_apps_user.c:3176: domain_deinit for domain 19: dev -1
calculator[46175]: fastrpc_apps_user.c:3176: domain_deinit for domain 20: dev -1
calculator[46175]: fastrpc_apps_user.c:3176: domain_deinit for domain 21: dev -1
calculator[46175]: fastrpc_apps_user.c:3176: domain_deinit for domain 22: dev -1
[... cut ...]
calculator[46175]: mod_table.c:220: open_mod_table_dtor_imp: closed reverse module apps_mem with handle 0x8b8d6c30
calculator[46175]: mod_table.c:220: open_mod_table_dtor_imp: closed reverse module apps_std with handle 0x8b8d6d30
calculator[46175]: fastrpc_context.c:325: fastrpc_context_table_deinit done
calculator[46175]: fastrpc_apps_user.c:4153: fastrpc_apps_user_deinit done
```

`adspmsgd not supported. nErr=8000043f` looks suspicious, this appears to be `AEE_ENOSUCHINSTANCE` and this appears not to happen in Android.

CPU: X1E78100 (Hexagon v73)
OS: Arch Linux ARM

anonymix007, 2025-02-17:
Hi!

You're right, this doesn't seem to happen without the `calculator.farf`, but I still can't see any DSP logs.
```
calculator[2043]: fastrpc_context.c:299: fastrpc_context_table_init done
calculator[2043]: fastrpc_config.c:380: Reading configuration file: calculator.debugconfig
calculator[2043]: fastrpc_config.c:418: fastrpc_config_init: Couldn't find file calculator.debugconfig, errno (No such file or directory) at ., /usr/lib/dsp//cdsp, /usr/lib/dsp/,
calculator[2043]: fastrpc_apps_user.c:4256: fastrpc_apps_user_init done. default domain:3 and &fastrpc_trace:0xffff9cb307a0
calculator[2043]: fastrpc_apps_user.c:4377: multidsplib_env_init: libcdsprpc.so loaded
calculator[2043]: fastrpc_apps_user.c:2703: remote_session_control Unsigned PD enable 1 request for domain 3
calculator[2043]: fastrpc_apps_user.c:2858: remote_session_control Register PD status notification request for domain 3
calculator[2043]: fastrpc_apps_user.c:3533: Successfully opened /usr/lib/dsp/cdsp/fastrpc_shell_unsigned_3, domain 3
calculator[2043]: fastrpc_apps_user.c:3641: Created user PD on domain 3, dbg_trace 0x0, enabled attr=> RPC timeout:0, Dbg Mode:N, CRC:N, Unsigned:Y, Signed:N, Adapt QOS:N, PD dump: (Config:N, Dbg:N), Perf: (Kernel:N, DSP:N), Iregion:N, QTF:N, UAF:N userPD initmem len:0x500000, Log pkt: N
calculator[2043]: fastrpc_apps_user.c:880: Warning 0xffffffff: get_lib_name failed for uri (null)
calculator[2043]: fastrpc_apps_user.c:3593: Error 0x14: fastrpc_enable_kernel_optimizations failed for domain 3 (Success)
calculator[2043]: fastrpc_perf.c:323: fastrpc_perf_init: enabled systrace 0x0 and RPC traces (kernel 0, dsp 0) with frequency 1000
calculator[2043]: fastrpc_apps_user.c:880: Warning 0xffffffff: get_lib_name failed for uri (null)
calculator[2043]: fastrpc_apps_user.c:880: Warning 0xffffffff: get_lib_name failed for uri (null)
calculator[2043]: log_config.c:396: file_watcher_thread starting for domain 3
calculator[2043]: log_config.c:420:file_watcher_thread: Couldn't find file calculator.farf, errno (No such file or directory) at ;/usr/lib/rfsa/adsp;/vendor/lib/rfsa/adsp;/vendor/dsp/;/usr/lib/dsp/;
calculator[2043]: fastrpc_apps_user.c:1535: Error 0x8000043e: remote_handle64_invoke failed for module (null), handle 0xffff94000b70, method 5 on domain 3 (sc 0x5000000) (errno Success)
calculator[2043]: listener_android.c:351: Error 0x8000043e: listener_start_thread domains support not available in listener
calculator[2043]: fastrpc_apps_user.c:1535: Error 0x8000043e: remote_handle64_invoke failed for module (null), handle 0xaaaacb548c40, method 2 on domain 3 (sc 0x2020200) (errno Success)
calculator[2043]: fastrpc_apps_user.c:1710: Warning 0x8000043e: remote_handle_open_domain: remotectl1 domains not supported for domain 3
calculator[2043]: listener_android.c:131: listener thread starting
calculator[2043]: fastrpc_apps_user.c:880: Warning 0xffffffff: get_lib_name failed for uri (null)
calculator[2043]: apps_std_imp.c:1035: Entering apps_std_fopen_with_env
calculator[2043]: apps_std_imp.c:1111: Successfully opened file libcalculator_skel.so
calculator[2043]: fastrpc_apps_user.c:880: Warning 0xffffffff: get_lib_name failed for uri (null)
calculator[2043]: fastrpc_apps_user.c:1836: remote_handle64_open: Successfully opened handle 0xaaaacb548d60 (remote 0x1ea9b90) for file:///libcalculator_skel.so?calculator_skel_handle_invoke&_modver=1.0&_idlver=1.2.3&_dom=cdsp on domain 3 (spawn time 20542 us, load time 9300 us), num handles 1
calculator[2043]: fastrpc_apps_user.c:1836: remote_handle64_open: Successfully opened handle 0xaaaacb548e80 (remote 0x1eaa910) for file:///libcalculator_skel.so?calculator_skel_handle_invoke&_modver=1.0&_idlver=1.2.3&_dom=cdsp on domain 3 (spawn time 1 us, load time 1359 us), num handles 2
calculator[2043]: fastrpc_apps_user.c:1889: remote_handle_close_domain: closed module with handle 0x1ea9b90 (skel unload time 1214 us)
calculator[2043]: fastrpc_apps_user.c:841: List of open handles on domain 3:
calculator[2043]: fastrpc_apps_user.c:846: libcalculator_skel.so, handle 0x1eaa910
calculator[2043]: fastrpc_apps_user.c:846: libcalculator_skel.so, handle 0x1ea9b90
calculator[2043]: fastrpc_apps_user.c:1959: remote_handle64_close: closed module libcalculator_skel.so with handle 0xaaaacb548d60 remote handle 0x1ea9b90, num of open handles: 1
calculator[2043]: fastrpc_apps_user.c:1959: remote_handle64_close: closed module libcalculator_skel.so with handle 0xaaaacb548e80 remote handle 0x1eaa910, num of open handles: 0
calculator[2043]: fastrpc_apps_user.c:3176: domain_deinit for domain 3: dev 6
calculator[2043]: listener_android.c:189::error: 107: AEE_SUCCESS == (nErr = fastrpc_session_get(domain))
calculator[2043]: fastrpc_notif.c:104: Warning: fastrpc_notif_domain_deinit: unable to find hash-node for domain 3
calculator[2043]: log_config.c:440:Warning: file_watcher_thread received exit for domain 3, file calculator.farf
calculator[2043]: log_config.c:513: file_watcher_thread exiting for domain 3
calculator[2043]: fastrpc_apps_user.c:3388: close_device_node: closed dev 6 on domain 3
calculator[2043]: fastrpc_apps_user.c:3225: domain_deinit: closed device 6 on domain 3 (kill time 2264 us)
calculator[2043]: fastrpc_apps_user.c:3890: close_dev: unloading library libcdsprpc.so
calculator[2043]: fastrpc_apps_user.c:3176: domain_deinit for domain 0: dev -1
calculator[2043]: fastrpc_apps_user.c:3176: domain_deinit for domain 1: dev -1
[... cut ...]
$ sudo ADSP_LIBRARY_PATH=. LD_LIBRARY_PATH=. ./calculator -d 3 -U 1 -n 1000000

Starting calculator test
Attempting to run on unsigned PD on domain 3

Allocate 4000000 bytes from ION heap
Creating sequence of numbers from 0 to 999999
Compute sum on domain 3

Call calculator_sum on the DSP
Sum = 499999500000

Call calculator_max on the DSP
Max value = 999999
Success
```

Another thing is that I tried to run with both `cdsprpcd` launched or not and this executable doesn't seem to do much. Seems to work the same in both cases.

quic-ekangupt, 2025-02-17:
Thanks for sharing the logs.

I see you have mentioned that calculator is working pre- and post- modification and you just don't see DSP logs in journalctl.

The reason is due to failure for adspmsgd init as you had mentioned earlier.

adspmsgd framework routes DSP messages to journalctl and if this framework is not enabled, DSP logs won't come in journalctl. You can still see the DSP logs in other diagnostic tools like QXDM or mini-dm(from Hexagon SDK). But to make routing work, we will need to fix adspmsgd problem.

Do you have dmesg logs for the adspmsgd failure? I believe there is some from fastrpc driver which is causing the apps_mem_request_mmap64 failures.

anonymix007, 2025-02-17:
Hmm, but I don't see any adspmsgd issues without the .farf file. Does it mean that adspmsgd isn't started without it or it's just not failing?

I'm getting this in dmesg each time I run the calculator example:
```
[12115.213643] qcom,fastrpc-cb 32300000.remoteproc:glink-edge:fastrpc:compute-cb@2: flag not supported 0x0
```

quic-ekangupt, 2025-02-17:
> Hmm, but I don't see any adspmsgd issues without the .farf file. Does it mean that adspmsgd isn't started without it or it's just not failing?
> 
Yes, adspmsgd is enabled only if .farf file is present. 
> I'm getting this in dmesg each time I run the calculator example:
> 
> ```
> [12115.213643] qcom,fastrpc-cb 32300000.remoteproc:glink-edge:fastrpc:compute-cb@2: flag not supported 0x0
> ```
This might be the reason for the failure. Driver is not supporting userspace allocated memory for mapping onto DSP.
I had raised patches to add support for this but had to drop it due to some reasons. I'll plan to resend it.
Ref: https://lore.kernel.org/all/20240508054250.2922-3-quic_ekangupt@quicinc.com/

anonymix007, 2025-02-17:
Okay, looks like I'd have to wait for you to resend this patch as it has too many conflicts.

anonymix007, 2025-02-17:
> You can still see the DSP logs in other diagnostic tools like QXDM or mini-dm(from Hexagon SDK)

Looks like these both are for remote debugging and I can't use them from the target. I guess I'll use `fopen` and `fwrite` for now.

quic-ekangupt, 2025-02-18:
> Okay, looks like I'd have to wait for you to resend this patch as it has too many conflicts.

Sure, I'll post an update here when I resend the patch

anonymix007, 2025-02-20:
Thanks for working on this!

Another thing is that there doesn't seem to be an easy way to use LLDB on target. While the `remote_debug_agent` exists, it needs access to some Android-specific paths (i.e. `/data`) and the `hexagon-lldb` executable can only by run via an emulator like [box64](https://github.com/ptitSeb/box64) which is considerably slower than natively.

Are there any plans to release Hexagon SDK for aarch64 Linux platform?

quic-ekangupt, 2025-02-25:
> Thanks for working on this!
>
> Another thing is that there doesn't seem to be an easy way to use LLDB on target. While the `remote_debug_agent` exists, it needs access to some Android-specific paths [...]

I'll check and get back on this.

anonymix007, 2025-03-31:
Any news on this issue? It's been more than a month

quic-asitsahoo, 2025-04-18:
Can you please let us know your concern on the access to the Android-specific paths? 
Apologies for the delay. We were having internal discussions within our team regarding this.
We don't have any plans to release the Hexagon SDK for aarch64 Linux platforms at the moment. However, we will keep you updated if there are any changes in the near future.

anonymix007, 2025-04-18:
I already don't remember what it was exactly, it's been two months. Maybe this:
```
$ strings remote_debug_agent/le/remote_debug_agent_cdsp | grep /data
/data/remote_debug_agent.pid
```
> We don't have any plans to release the Hexagon SDK for aarch64 Linux platforms at the moment. However, we will keep you updated if there are any changes in the near future.

That's unfortunate. Snapdragon laptops running Linux are the best machines for Hexagon development. It's such a shame that this use-case is not supported. And now I'm kind of stuck without any realistic way to debug crashes.

anonymix007, 2025-06-03:
@quic-ekangupt are there any news on adspmsgd fixes? It's been more than 3 months since this got originally reported.

quic-ekangupt, 2025-07-04:
Hi @anonymix007,

Apologies for the delay, there were some discussions happening around development on fastrpc.

I've raised a patch for fixing the apps_mem failure. Please find the patch here:
https://lore.kernel.org/all/20250704083726.1901705-1-ekansh.gupta@oss.qualcomm.com/

Let me know if you are facing any problems with this.

quic-ekangupt, 2025-07-08:
> Thanks, logs seem to be working with your patch.
> 
> This is (a part of) what I get as output when I run my program:
> 
> ```
> adsprpc:dsp: 1b9e:304: CDSP0: [DSP]: Calling get_static_properties (200fc80)...
> 
> adsprpc:dsp: 1b9e:304: CDSP0: abort
> adsprpc:dsp: 1b9e:304: CDSP0:  -- terminating
> 
> [CPU]: Failed to load sbc_enc_module.so.1: 0x8000040d
> ```
> 
> Lines with `[CPU]` and `[DSP]` substrings are my own logs, others, I assume, are from the firmware.
> 
> However, I don't see the crash backtrace. It was present on Android by default without any extra setup. Do you have any suggestions on how to get backtrace here?

Is it possible for you to share complete logs? The expectation is that crash signature will be seen after you pick above patch, without any additional steps. I'm seeing "abort" and "termination" followed by 0x8000040d(which suggests that the DSP process is exiting) but I couldn't see any reason for this.

ekanshibu, 2025-09-08:
Hi @anonymix007 ,

We have recently fixed some issues as well as added few tests to this project. Can you please try out running fastrpc_test from this project? From the logs shared I don't see any concerning problems. Is it possible for me to run the same use case at my end?

anonymix007, 2025-09-15:
Hi @quic-parihar 

I'm using Arch Linux Arm with a custom kernel: https://github.com/anonymix007/linux-x1e/tree/wip/x1e80100-6.16-rc4
My device is Lenovo T14s Gen 6 (X1E781000).

quic-parihar, 2025-09-16:
Hi @anonymix007,

I had PD exception logging enabled in my local workspace, which caused the crash call trace to appear in the journalctl logs.

I've updated the PR to enable PD exception logging by default:
🔗 fastrpc PR: #243 https://github.com/qualcomm/fastrpc/pull/243

This change, along with the patch from the kernel mailing list:
🔗 Lore Patch: https://lore.kernel.org/all/20250704083726.1901705-1-ekansh.gupta@oss.qualcomm.com/
should help improve visibility of crash messages in the logs.

quic-vkatoch, 2026-01-02:
Closing this issue for now. Please reopen if needed.