qualcomm/fastrpc #144: Behaviour when /dev/fastrpc-adsp-secure file is not present

laklimov, 2025-02-19:
When adsprpcd starts it tries to use file `/dev/fastrpc-adsp-secure`. If this file is not present then maybe it makes sense to fallback to non-secure version of that dev file i.e. `/dev/fastrpc-adsp`. Currently adsprpcd doesn't do it and constantly tries to restart.

It will be good to have feature to pass it as an argument for instance. Also, after failing adsprpcd tries to restart constantly. What do you think about giving up after some number of attempts? It pollutes logs too.

quic-bkumar, 2025-04-17:
Pls feel free to drop in a patch for this, will review and approve it.

quic-ekangupt, 2025-04-23:
Issue is closed with this PR https://github.com/quic/fastrpc/pull/157

closed 2025-04-23 by quic-ekangupt