qualcomm/fastrpc #152: Incomplete DCO in commits

krzk, 2025-03-13:
Contributing guide claims you must follow "Developer Certificate of Origin" but commits do not - for several commits there is no Signed-off-by by the committer. Example commit 7c4345e139774e3a30d50b1cac09902fdd4ed91c - missing SoB from Ekansh

[image]

If you require DCO then entire commit chain must have proper certifications. Otherwise what is the point of author certifying the origin if committer can change/add anything, including non-GPL compliant code, and still not sign the DCO?

quic-ekangupt, 2025-04-22:
Thanks Krzysztof for pointing this out.

I'm working on fixing this.

quic-ekangupt, 2025-04-23:
Looks like when the maintainers are merging the changes, the committer is getting modified to maintainer. I'm checking how to avoid this.

krzk, 2025-04-24:
No. Merging does not change committer. Rebasing does. Github is just an interface to Git, so all standard Git rules apply. Solutions are: merge instead of rebase or rebase in Git with signining off (not Github) or drop DCO.

quic-ekangupt, 2025-05-16:
I am working on enabling merge by the PR owners with some ruleset enforced. Basically the plan is that PR owners will be able to merge/auto-merge the changes once 2 approvals are received. I hope this will resolve the problem.