qualcomm/fastrpc #166: soname of the library in tests doesn't match build libraries

lumag, 2025-04-24:
libraries under `test/linux` have `libadsprpc.so` listed as required library, however built library provides a different SONAME. In OE this results in QA error. Please fix the SONAME in the provided libraries:

```
$ readelf -d test/linux/libcalculator.so

Dynamic section at offset 0x4e08 contains 22 entries:
  Tag        Type                         Name/Value
 0x0000000000000001 (NEEDED)             Shared library: [libadsprpc.so]
 0x0000000000000001 (NEEDED)             Shared library: [libm.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libstdc++.so.6]
 0x000000000000000e (SONAME)             Library soname: [libcalculator.so]
 0x0000000000000004 (HASH)               0x1b8
 0x0000000000000005 (STRTAB)             0x890
 0x0000000000000006 (SYMTAB)             0x338
 0x000000000000000a (STRSZ)              838 (bytes)
 0x000000000000000b (SYMENT)             24 (bytes)
 0x0000000000000003 (PLTGOT)             0x14fe8
 0x0000000000000002 (PLTRELSZ)           816 (bytes)
 0x0000000000000014 (PLTREL)             RELA
 0x0000000000000017 (JMPREL)             0xef0
 0x0000000000000007 (RELA)               0xc68
 0x0000000000000008 (RELASZ)             648 (bytes)
 0x0000000000000009 (RELAENT)            24 (bytes)
 0x000000006ffffffe (VERNEED)            0xc48
 0x000000006fffffff (VERNEEDNUM)         1
 0x000000006ffffff0 (VERSYM)             0xbd6
 0x000000006ffffff9 (RELACOUNT)          20
 0x0000000000000000 (NULL)               0x0
$ readelf -d build/src/.libs/libadsprpc.so

Dynamic section at offset 0x6fd00 contains 26 entries:
  Tag        Type                         Name/Value
 0x0000000000000001 (NEEDED)             Shared library: [libm.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
 0x000000000000000e (SONAME)             Library soname: [libadsprpc.so.1]
 0x000000000000000c (INIT)               0xe658
 0x000000000000000d (FINI)               0x4830c
 0x0000000000000019 (INIT_ARRAY)         0x7d100
 0x000000000000001b (INIT_ARRAYSZ)       32 (bytes)
 0x000000000000001a (FINI_ARRAY)         0x7d120
 0x000000000000001c (FINI_ARRAYSZ)       16 (bytes)
 0x000000006ffffef5 (GNU_HASH)           0x1f0
 0x0000000000000005 (STRTAB)             0x4a00
 0x0000000000000006 (SYMTAB)             0xfc8
 0x000000000000000a (STRSZ)              11362 (bytes)
 0x000000000000000b (SYMENT)             24 (bytes)
 0x0000000000000003 (PLTGOT)             0x7ffe8
 0x0000000000000002 (PLTRELSZ)           11352 (bytes)
 0x0000000000000014 (PLTREL)             RELA
 0x0000000000000017 (JMPREL)             0xba00
 0x0000000000000007 (RELA)               0x7bc0
 0x0000000000000008 (RELASZ)             15936 (bytes)
 0x0000000000000009 (RELAENT)            24 (bytes)
 0x000000006ffffffe (VERNEED)            0x7b40
 0x000000006fffffff (VERNEEDNUM)         2
 0x000000006ffffff0 (VERSYM)             0x7662
 0x000000006ffffff9 (RELACOUNT)          623
 0x0000000000000000 (NULL)               0x0
```

OE error:
```
ERROR: fastrpc-0.1+-r0 do_package_qa: QA Issue: /usr/lib/fastrpc_test/libmultithreading.so contained in package fastrpc-test requires libadsprpc.so()(64bit), but no providers found in RDEPENDS:fastrpc-test? [file-rdeps]
ERROR: fastrpc-0.1+-r0 do_package_qa: Fatal QA errors were found, failing task.
```

quic-ekangupt, 2025-04-25:
Thanks for raising this issue. As there is a symlink created between un-versioned and versioned library, we are not observing any runtime problems. 

Vinayak will be working on updating the libs with proper linking.

Also, can you please help with the steps to run this QA? @quic-mtharu might be able to include it to our CI.

lumag, 2025-04-25:
> Thanks for raising this issue. As there is a symlink created between un-versioned and versioned library, we are not observing any runtime problems.
>
> Vinayak will be working on updating the libs with [...]

It's a standard OpenEmbedded QA, once I added test files to the package.

quic-vkatoch, 2025-04-30:
I have addressed the changes with this PR - https://github.com/quic/fastrpc/pull/170

closed 2025-05-16 by quic-bkumar