qualcomm/fastrpc #167: RB3: error starting the daemom

lumag, 2025-04-24:
On the RB3 board running cdsprpcd results in a stream of errors:

```
Apr 03 20:39:33 qcom-armv8a cdsprpcd[481]: ../../git/src/rpcmem_linux.c:95: Error 2: Unable to open /dev/dma_heap/system
Apr 03 20:39:33 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_context.c:299: fastrpc_context_table_init done
Apr 03 20:39:33 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_config.c:380: Reading configuration file: cdsprpcd.debugconfig
Apr 03 20:39:34 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_config.c:418: fastrpc_config_init: Couldn't find file cdsprpcd.debugconfig, errno (No such file or directory) at /usr/share/qcom/sdm845/Thundercomm/db845c/dsp//cdsp, /usr/share/qcom/sdm845/Thundercomm/db845c/dsp/,
Apr 03 20:39:34 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:4258: fastrpc_apps_user_init done. default domain:3 and &fastrpc_trace:0xffff89ef1660
Apr 03 20:39:34 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:4379: multidsplib_env_init: libcdsprpc.so loaded
Apr 03 20:39:34 qcom-armv8a cdsprpcd[481]: ../../git/src/cdsprpcd.c:43:cdsp_default_listener_start called
Apr 03 20:39:34 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:880: Warning 0x0: get_lib_name failed for uri '":;./\attachguestos
Apr 03 20:39:34 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:1784: remote_handle_open: Successfully opened handle 0x103 for '":;./\attachguestos on domain 3 (spawn time 0 us, load time 0 us)
Apr 03 20:39:34 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_cap.c:133: Warning 0xffffffff: fastrpc_get_cap failed to get attribute 131 for domain 3 (errno Operation not permitted)
Apr 03 20:39:34 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:880: Warning 0xffffffff: get_lib_name failed for uri (null)
Apr 03 20:39:34 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_perf.c:323: fastrpc_perf_init: enabled systrace 0x0 and RPC traces (kernel 0, dsp 0) with frequency 1000
Apr 03 20:39:34 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_cap.c:133: Warning 0xffffffff: fastrpc_get_cap failed to get attribute 129 for domain 3 (errno Operation not permitted)
Apr 03 20:39:34 qcom-armv8a cdsprpcd[481]: ../../git/src/log_config.c:396: file_watcher_thread starting for domain 3
Apr 03 20:39:34 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:880: Warning 0xffffffff: get_lib_name failed for uri (null)
Apr 03 20:39:34 qcom-armv8a cdsprpcd[481]: ../../git/src/log_config.c:420:file_watcher_thread: Couldn't find file cdsprpcd.farf, errno (No such file or directory) at /usr/share/qcom/sdm845/Thundercomm/db845c/dsp/
Apr 03 20:39:34 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:1535: Error 0x4e: remote_handle64_invoke failed for module (null), handle 0xb7a3c10, method 2 on domain 3 (sc 0x2020200) (errno Success)
Apr 03 20:39:34 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:880: Warning 0x0: get_lib_name failed for uri adsp_default_listener
Apr 03 20:39:34 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:1784: remote_handle_open: Successfully opened handle 0xffffffff for adsp_default_listener on domain 3 (spawn time 484167 us, load time 366579 us)
Apr 03 20:39:34 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:880: Warning 0xffffffff: get_lib_name failed for uri (null)
Apr 03 20:39:34 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:1535: Error 0x4e: remote_handle64_invoke failed for module (null), handle 0xb7a3c10, method 2 on domain 3 (sc 0x2020200) (errno Operation not permitted)
Apr 03 20:39:34 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:880: Warning 0x0: get_lib_name failed for uri adsp_default_listener
Apr 03 20:39:34 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:1784: remote_handle_open: Successfully opened handle 0xffffffff for adsp_default_listener on domain 3 (spawn time 3 us, load time 2670 us)
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:1535: Error 0x4e: remote_handle64_invoke failed for module (null), handle 0xffff7c000bf0, method 5 on domain 3 (sc 0x5000000) (errno Operation not permitted)
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/listener_android.c:131: listener thread starting
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/adsp_default_listener.c:364::error: 44: 0 == (nErr = adsp_default_listener_register())
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/adsp_default_listener.c:428:Error 0x2c: adsp_default_listener_start exiting
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:3892: close_dev: unloading library libcdsprpc.so
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:3178: domain_deinit for domain 0: dev -1
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:3178: domain_deinit for domain 1: dev -1
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:3178: domain_deinit for domain 2: dev -1
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:3178: domain_deinit for domain 3: dev 4
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/listener_android.c:189::error: 107: AEE_SUCCESS == (nErr = fastrpc_session_get(domain))
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/listener_android.c:374:Error 0x4e: listener_start_thread failed for domain 3
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_notif.c:104: Warning: fastrpc_notif_domain_deinit: unable to find hash-node for domain 3
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/log_config.c:440:Warning: file_watcher_thread received exit for domain 3, file cdsprpcd.farf
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/log_config.c:513: file_watcher_thread exiting for domain 3
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:3390: close_device_node: closed dev 4 on domain 3
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:3227: domain_deinit: closed device 4 on domain 3 (kill time 2580 us)
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:3178: domain_deinit for domain 4: dev -1
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:3178: domain_deinit for domain 5: dev -1
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:3178: domain_deinit for domain 6: dev -1
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:3178: domain_deinit for domain 7: dev -1
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:3178: domain_deinit for domain 8: dev -1
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:3178: domain_deinit for domain 9: dev -1
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:3178: domain_deinit for domain 10: dev -1
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:3178: domain_deinit for domain 11: dev -1
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:3178: domain_deinit for domain 12: dev -1
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:3178: domain_deinit for domain 13: dev -1
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:3178: domain_deinit for domain 14: dev -1
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:3178: domain_deinit for domain 15: dev -1
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:3178: domain_deinit for domain 16: dev -1
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:3178: domain_deinit for domain 17: dev -1
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:3178: domain_deinit for domain 18: dev -1
Apr 03 20:39:35 qcom-armv8a cdsprpcd[481]: ../../git/src/fastrpc_apps_user.c:3178: domain_deinit for domain 19: dev -1
```

The daemon then restarts itself continuously.

lumag, 2025-04-24:
Enabling `CONFIG_DMABUF_HEAPS_SYSTEM=y` doesn't fix the issue.

lumag, 2025-04-24:
For the reference, the daemon was started as:
```sh
env DSP_LIBRARY_PATH=/usr/share/qcom/sdm845/Thundercomm/db845c/dsp cdsprpc
```

lumag, 2025-04-24:
With DMABUF_HEAPS:

```
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_context.c:299: fastrpc_context_table_init done
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_config.c:380: Reading configuration file: cdsprpcd.debugconfig
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_config.c:418: fastrpc_config_init: Couldn't find file cdsprpcd.debugconfig, errno (No such file or directory) at /usr/share/qcom/sdm845/Thundercomm/db845c/dsp//cdsp, /usr/share/qcom/sdm845/Thundercomm/db845c/dsp/,
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_apps_user.c:4258: fastrpc_apps_user_init done. default domain:3 and &fastrpc_trace:0xffffa6d31660
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_apps_user.c:4379: multidsplib_env_init: libcdsprpc.so loaded
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/cdsprpcd.c:43:cdsp_default_listener_start called
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_apps_user.c:880: Warning 0x0: get_lib_name failed for uri '":;./\attachguestos
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_apps_user.c:1784: remote_handle_open: Successfully opened handle 0x103 for '":;./\attachguestos on domain 3 (spawn time 0 us, load time 0 us)
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_cap.c:133: Warning 0xffffffff: fastrpc_get_cap failed to get attribute 131 for domain 3 (errno Operation not permitted)
Apr 03 20:35:03 qcom-armv8a kernel: qcom,fastrpc remoteproc-cdsp:glink-edge.fastrpcglink-apps-dsp.-1.-1: Error: dsp information is incorrect err: -1
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_apps_user.c:880: Warning 0xffffffff: get_lib_name failed for uri (null)
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_perf.c:323: fastrpc_perf_init: enabled systrace 0x0 and RPC traces (kernel 0, dsp 0) with frequency 1000
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/log_config.c:396: file_watcher_thread starting for domain 3
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/log_config.c:420:file_watcher_thread: Couldn't find file cdsprpcd.farf, errno (No such file or directory) at /usr/share/qcom/sdm845/Thundercomm/db845c/dsp/
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_cap.c:133: Warning 0xffffffff: fastrpc_get_cap failed to get attribute 129 for domain 3 (errno Operation not permitted)
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_apps_user.c:880: Warning 0xffffffff: get_lib_name failed for uri (null)
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_apps_user.c:880: Warning 0xffffffff: get_lib_name failed for uri (null)
Apr 03 20:35:03 qcom-armv8a kernel: qcom,fastrpc remoteproc-cdsp:glink-edge.fastrpcglink-apps-dsp.-1.-1: Error: dsp information is incorrect err: -1
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_apps_user.c:1535: Error 0x4e: remote_handle64_invoke failed for module (null), handle 0xffff98000eb0, method 5 on domain 3 (sc 0x5000000) (errno Success)
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/listener_android.c:131: listener thread starting
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_apps_user.c:1535: Error 0x4e: remote_handle64_invoke failed for module (null), handle 0xffff98000cf0, method 2 on domain 3 (sc 0x2020200) (errno Operation not permitted)
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_apps_user.c:880: Warning 0x0: get_lib_name failed for uri adsp_default_listener
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_apps_user.c:1784: remote_handle_open: Successfully opened handle 0xffffffff for adsp_default_listener on domain 3 (spawn time 110152 us, load time 2491 us)
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_apps_user.c:1535: Error 0x4e: remote_handle64_invoke failed for module (null), handle 0xffff98000cf0, method 2 on domain 3 (sc 0x2020200) (errno Operation not permitted)
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_apps_user.c:880: Warning 0x0: get_lib_name failed for uri adsp_default_listener
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_apps_user.c:1784: remote_handle_open: Successfully opened handle 0xffffffff for adsp_default_listener on domain 3 (spawn time 1 us, load time 1136 us)
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/adsp_default_listener.c:364::error: 44: 0 == (nErr = adsp_default_listener_register())
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/adsp_default_listener.c:428:Error 0x2c: adsp_default_listener_start exiting
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_apps_user.c:3892: close_dev: unloading library libcdsprpc.so
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_apps_user.c:3178: domain_deinit for domain 0: dev -1
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_apps_user.c:3178: domain_deinit for domain 1: dev -1
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_apps_user.c:3178: domain_deinit for domain 2: dev -1
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_apps_user.c:3178: domain_deinit for domain 3: dev 5
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_apps_user.c:1491: Error 0x4e: remote_handle_invoke failed for handle 0x3, method 4 on domain 3 (sc 0x4020200) (errno Operation not permitted)
[... cut ...]
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_apps_user.c:3178: domain_deinit for domain 29: dev -1
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_apps_user.c:3178: domain_deinit for domain 30: dev -1
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_apps_user.c:3178: domain_deinit for domain 31: dev -1
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_context.c:325: fastrpc_context_table_deinit done
Apr 03 20:35:03 qcom-armv8a cdsprpcd[441]: ../../git/src/fastrpc_apps_user.c:4155: fastrpc_apps_user_deinit done
```

quic-ekangupt, 2025-04-25:
This issue looks to be coming due to bad handle.

I have tested meta-qcom + (github switch PR) on RB3Gen2 board and there I didn't see any such issue. Can you please suggest me how can I setup my board to reproduce this issue? Need to just understand how it was not seen at my end.

lumag, 2025-04-25:
> This issue looks to be coming due to bad handle.
> 
> I have tested meta-qcom + (github switch PR) on RB3Gen2 board and there I didn't see any such issue. Can you please suggest me how can I setup my board to reproduce this issue? Need to just understand how it was not seen at my end.

It's RB3, not RB3 Gen2.

quic-ekangupt, 2025-04-28:
I see handle 0xffffffff is causing the problem and I'm able to see a scenario which might cause this problem. I don't have access to RB3 board as of now. Trying to find a board to experiment it out from my end.

quic-ekangupt, 2025-04-30:
I've acquired an RB5 board and will test the changes on it. I'll provide updates soon. Thanks.

quic-ekangupt, 2025-05-19:
Tested the change: https://github.com/quic/fastrpc/pull/174 on older platforms and daemon is working with this.

closed 2025-05-21 by quic-ekangupt