qualcomm/fastrpc #176: Failing tests on X13s

lumag, 2025-05-23:
On the Lenovo X13s several tests fail:

```
Allocate 4000 bytes from ION heap
Creating sequence of numbers from 0 to 999
Compute sum on domain 3

Call calculator_sum on the DSP
Sum = 499500

Call calculator_max on the DSP
Max value = 999
Success in /home/lumag/Projects/Qcomm/Tools/fastrpc-quic/test/linux/libcalculator.so
Test PASSED
Please look at the mini-dm logs or the adb logcat logs for DSP output
Success in /home/lumag/Projects/Qcomm/Tools/fastrpc-quic/test/linux/libmultithreading.so

Demonstrating HAP_compute_res.h APIs

hap_example function PASSED
Please look at the mini-dm logs or the adb logcat logs for DSP output

Demonstrating FARF run-time logging

hap_example function PASSED
Please look at the mini-dm logs or the adb logcat logs for DSP output

Demonstrating HAP_mem.h APIs

hap_example function PASSED
Please look at the mini-dm logs or the adb logcat logs for DSP output

Demonstrating HAP_perf.h APIs

hap_example function PASSED
Please look at the mini-dm logs or the adb logcat logs for DSP output

Demonstrating HAP_power.h APIs

hap_example function PASSED
Please look at the mini-dm logs or the adb logcat logs for DSP output
ERROR: sysmon_cachelock.h APIs supported only on signed CDSP

hap_example function FAILED
Please look at the mini-dm logs or the adb logcat logs for DSP output
Error occurred with selector 5: 57

Demonstrating HAP_vtcm_mgr.h APIs

hap_example function PASSED
Please look at the mini-dm logs or the adb logcat logs for DSP output
The PD type is 1 (unsigned)
hap_example function PASSED
Please look at the mini-dm logs or the adb logcat logs for DSP output

Demonstrating HAP_compute_res.h v2 APIs

hap_example function FAILED
Please look at the mini-dm logs or the adb logcat logs for DSP output
Error occurred with selector 10: 78
Test failed with error code 0x4e in /home/lumag/Projects/Qcomm/Tools/fastrpc-quic/test/linux/libhap_example.so
Test failed with error code 0x4e
```

lumag, 2025-05-23:
Note, testing domain 2 (SDSP) fails completely:

```
# LD_LIBRARY_PATH=../build/src/.libs/ ../build/test/fastrpc_test  -d 2

Allocate 4000 bytes from ION heap
Creating sequence of numbers from 0 to 999
Compute sum on domain 2
Retry attempt unsuccessful. Timing out....
ERROR 0x27: Failed to compute sum on domain 2
ERROR 0x27: Failed to find max on domain 2
Test failed with error code 0x27 in /home/lumag/Projects/Qcomm/Tools/fastrpc-quic/test/lin2/libcalculator.so
Test failed with error code 0x27
```

```
2025-05-23T15:21:27.282842+03:00 rohan fastrpc_test: ../../src/fastrpc_context.c:299: fastrpc_context_table_init done
2025-05-23T15:21:27.283355+03:00 rohan fastrpc_test: ../../src/fastrpc_config.c:380: Reading configuration file: fastrpc_test.debugconfig
2025-05-23T15:21:27.283407+03:00 rohan fastrpc_test: ../../src/fastrpc_config.c:418: fastrpc_config_init: Couldn't find file fastrpc_test.debugconfig, errno (No such file or directory) at /home/lumag/Projects/Qcomm/Tools/fastrpc-quic/test/v68, /vendor/dsp//cdsp, /vendor/dsp/, 
2025-05-23T15:21:27.283453+03:00 rohan fastrpc_test: ../../src/fastrpc_apps_user.c:4254: fastrpc_apps_user_init done. default domain:3 and &fastrpc_trace:0xffffb2551660
2025-05-23T15:21:27.283489+03:00 rohan fastrpc_test: ../../src/fastrpc_apps_user.c:4375: multidsplib_env_init: libcdsprpc.so loaded
2025-05-23T15:21:27.283515+03:00 rohan fastrpc_test: ../../src/fastrpc_apps_user.c:2701: remote_session_control Unsigned PD enable 1 request for domain 3
2025-05-23T15:21:27.283675+03:00 rohan fastrpc_test: ../../src/fastrpc_apps_user.c:3531: Successfully opened /vendor/dsp/cdsp/fastrpc_shell_unsigned_3, domain 3
2025-05-23T15:21:27.301472+03:00 rohan fastrpc_test: ../../src/fastrpc_apps_user.c:3639: Created user PD on domain 3, dbg_trace 0x0, enabled attr=> RPC timeout:0, Dbg Mode:N, CRC:N, Unsigned:Y, Signed:N, Adapt QOS:N, PD dump: (Config:N, Dbg:N), Perf: (Kernel:N, DSP:N), Iregion:N, QTF:N, UAF:N userPD initmem len:0x500000, Log pkt: N
2025-05-23T15:21:27.301673+03:00 rohan fastrpc_test: ../../src/fastrpc_apps_user.c:879: Warning 0xffffffff: get_lib_name failed for uri (null)
2025-05-23T15:21:27.301799+03:00 rohan fastrpc_test: ../../src/fastrpc_apps_user.c:3591: Error 0x14: fastrpc_enable_kernel_optimizations failed for domain 3 (Success)
2025-05-23T15:21:27.301907+03:00 rohan fastrpc_test: ../../src/fastrpc_perf.c:323: fastrpc_perf_init: enabled systrace 0x0 and RPC traces (kernel 0, dsp 0) with frequency 1000
2025-05-23T15:21:27.302015+03:00 rohan fastrpc_test: ../../src/fastrpc_apps_user.c:879: Warning 0xffffffff: get_lib_name failed for uri (null)
2025-05-23T15:21:27.302089+03:00 rohan fastrpc_test: ../../src/log_config.c:396: file_watcher_thread starting for domain 3
2025-05-23T15:21:27.302192+03:00 rohan fastrpc_test: ../../src/log_config.c:420:file_watcher_thread: Couldn't find file fastrpc_test.farf, errno (No such file or directory) at /home/lumag/Projects/Qcomm/Tools/fastrpc-quic/test/v68
2025-05-23T15:21:27.302281+03:00 rohan fastrpc_test: ../../src/fastrpc_apps_user.c:879: Warning 0xffffffff: get_lib_name failed for uri (null)
2025-05-23T15:21:27.304728+03:00 rohan fastrpc_test: ../../src/fastrpc_apps_user.c:1534: Error 0x80000600: remote_handle64_invoke failed for module (null), handle 0xaaaab2fc1270, method 2 on domain 3 (sc 0x2020200) (errno Success)
2025-05-23T15:21:27.304958+03:00 rohan fastrpc_test: ../../src/fastrpc_apps_user.c:1708: Warning 0x80000600: remote_handle_open_domain: remotectl1 domains not supported for domain 3
2025-05-23T15:21:27.305083+03:00 rohan fastrpc_test: ../../src/fastrpc_apps_user.c:1534: Error 0x80000600: remote_handle64_invoke failed for module (null), handle 0xffffa4000b70, method 5 on domain 3 (sc 0x5000000) (errno Success)
2025-05-23T15:21:27.305238+03:00 rohan fastrpc_test: ../../src/listener_android.c:351: Error 0x80000600: listener_start_thread domains support not available in listener
2025-05-23T15:21:27.305445+03:00 rohan fastrpc_test: ../../src/listener_android.c:131: listener thread starting
2025-05-23T15:21:27.305574+03:00 rohan fastrpc_test: ../../src/fastrpc_apps_user.c:879: Warning 0xffffffff: get_lib_name failed for uri (null)
2025-05-23T15:21:27.305719+03:00 rohan fastrpc_test: ../../src/apps_std_imp.c:1035: Entering apps_std_fopen_with_env
2025-05-23T15:21:27.305976+03:00 rohan fastrpc_test: ../../src/apps_std_imp.c:1111: Successfully opened file libcalculator_skel.so
2025-05-23T15:21:27.306132+03:00 rohan fastrpc_test: ../../src/fastrpc_apps_user.c:879: Warning 0xffffffff: get_lib_name failed for uri (null)
2025-05-23T15:21:27.309217+03:00 rohan fastrpc_test: ../../src/fastrpc_apps_user.c:1834: remote_handle64_open: Successfully opened handle 0xaaaab2fc1270 (remote 0xea4a60) for file:///libcalculator_skel.so?calculator_skel_handle_invoke&_modver=1.0&_idlver=1.2.3&_dom=cdsp on domain 3 (spawn time 18824 us, load time 7549 us), num handles 1
2025-05-23T15:21:27.309571+03:00 rohan fastrpc_test: ../../src/fastrpc_apps_user.c:1834: remote_handle64_open: Successfully opened handle 0xaaaab2fc75e0 (remote 0xea5520) for file:///libcalculator_skel.so?calculator_skel_handle_invoke&_modver=1.0&_idlver=1.2.3&_dom=cdsp on domain 3 (spawn time 1 us, load time 139 us), num handles 2
2025-05-23T15:21:27.309902+03:00 rohan fastrpc_test: ../../src/fastrpc_apps_user.c:1886: remote_handle_close_domain: closed module with handle 0xea4a60 (skel unload time 143 us)
[... cut ...]
2025-05-23T15:21:34.484442+03:00 rohan fastrpc_test: ../../src/fastrpc_apps_user.c:3174: domain_deinit for domain 29: dev -1
2025-05-23T15:21:34.484519+03:00 rohan fastrpc_test: ../../src/fastrpc_apps_user.c:3174: domain_deinit for domain 30: dev -1
2025-05-23T15:21:34.484589+03:00 rohan fastrpc_test: ../../src/fastrpc_apps_user.c:3174: domain_deinit for domain 31: dev -1
2025-05-23T15:21:34.484684+03:00 rohan fastrpc_test: ../../src/fastrpc_context.c:325: fastrpc_context_table_deinit done
2025-05-23T15:21:34.484782+03:00 rohan fastrpc_test: ../../src/fastrpc_apps_user.c:4151: fastrpc_apps_user_deinit done
```

quic-parihar, 2025-05-28:
I am looking into this issue. Will get back.

quic-parihar, 2025-07-10:
@lumag can you try with this PR https://github.com/qualcomm/fastrpc/pull/184.

We have tried on some devices with SLPI, where its working.

lumag, 2025-07-10:
Next week, I am out of office this week.

quic-ekangupt, 2025-08-05:
As per comment here, tests are now working on X13s:
https://github.com/qualcomm/fastrpc/pull/184#issuecomment-3096058526

Closing this issue.

closed 2025-08-05 by quic-ekangupt