qualcomm/fastrpc #188: ERROR 0x8000054f / "No context ID matches response" in dmesg on X1E-80-100 Linux

valpackett, 2025-06-26:
Just running the provided test program (Dell Latitude 7455, Linux 6.16.0-rc1-next-20250612 / postmarketOS Trailblazer)

```
$ doas env LD_LIBRARY_PATH=$PWD/../src/.libs ./fastrpc_test -a v75
LD_LIBRARY_PATH: /home/val/src/fastrpc/test/linux
DSP_LIBRARY_PATH: /home/val/src/fastrpc/test/v75
abs_lib_path: /home/val/src/fastrpc/test/linux
opening /home/val/src/fastrpc/test/linux/libcalculator.so: 0xaaaae42a55f0

Allocate 4000 bytes from ION heap
Creating sequence of numbers from 0 to 999
Compute sum on domain 3
Retry attempt unsuccessful. Timing out....
ERROR 0x8000054f: Failed to compute sum on domain 3
ERROR 0x8000054f: Failed to find max on domain 3
Test failed with error code 0x8000054f in /home/val/src/fastrpc/test/linux/libcalculator.so
opening /home/val/src/fastrpc/test/linux/libhap_example.so: 0xaaaae42a55b0
ERROR 0x8000054f: Unable to create FastRPC session on domain 3

hap_example function FAILED
Please look at the mini-dm logs or the adb logcat logs for DSP output
Error occurred with selector 0: -2147482289
ERROR 0x8000054f: Unable to create FastRPC session on domain 3

…[and so on]…
```

(the initial prints are extra debug prints because initially I've had an issue with library paths being cut off, had to workaround:)

```diff
--- i/src/fastrpc_procbuf.c
+++ w/src/fastrpc_procbuf.c
@@ -74,8 +74,8 @@ static int get_non_preload_lib_names (char** lib_names, size_t* buffer_size, int
        size_t dsp_search_path_len = std_strlen(DSP_LIBRARY_PATH) + 1;
 
        VERIFYC(*lib_names != NULL, AEE_ENOMEMORY);
-       VERIFYC(NULL != (data_paths = calloc(1, sizeof(char) * dsp_search_path_len)), AEE_ENOMEMORY);
-       VERIFYC(AEE_SUCCESS == apps_std_getenv(DSP_LIBRARY_PATH, data_paths, dsp_search_path_len, &env_list_len), AEE_EGETENV);
+       VERIFYC(NULL != (data_paths = calloc(1, sizeof(char) * dsp_search_path_len * 100)), AEE_ENOMEMORY);
+       VERIFYC(AEE_SUCCESS == apps_std_getenv(DSP_LIBRARY_PATH, data_paths, dsp_search_path_len * 100, &env_list_len), AEE_EGETENV);
 
        char* path = strtok_r(data_paths, ";", &saveptr);
        while (path != NULL)
```

While in the dmesg I see these lines (one per test basically):

```dmesg
[645307.117292] qcom,fastrpc 32300000.remoteproc:glink-edge.fastrpcglink-apps-dsp.-1.-1: No context ID matches response
[645307.123584] qcom,fastrpc 32300000.remoteproc:glink-edge.fastrpcglink-apps-dsp.-1.-1: No context ID matches response
[645307.130118] qcom,fastrpc 32300000.remoteproc:glink-edge.fastrpcglink-apps-dsp.-1.-1: No context ID matches response
…[and so on]…
```

(Kernel bug??)

valpackett, 2025-06-26:
<details>
<summary>More detailed log with farf messages and stuff (changed to just print to stderr)</summary>

```
LD_LIBRARY_PATH: /home/val/src/fastrpc/test/linux
DSP_LIBRARY_PATH: /home/val/src/fastrpc/test/v75
abs_lib_path: /home/val/src/fastrpc/test/linux
platform_libs.c:23: calling init for fastrpc_apps_user
platform_libs.c:23: calling init for gpls
platform_libs.c:25: init for gpls returned 0
platform_libs.c:23: calling init for rpcmem
platform_libs.c:25: init for rpcmem returned 0
fastrpc_context.c:299: fastrpc_context_table_init done
fastrpc_config.c:380: Reading configuration file: fastrpc_test.debugconfig

fastrpc_apps_user.c:get_current_domain: tlsKey == INVALID_KEY
apps_std_imp.c:1307: Entering apps_std_get_search_paths_with_env
apps_std_imp.c:941: Entering get_dirlist_from_env
apps_std_imp.c:1016: Exiting get_dirlist_from_env for DSP_LIBRARY_PATH, err 0
apps_std_imp.c:1366: Exiting apps_std_get_search_paths_with_env for envvarname ADSP_LIBRARY_PATH delim ;, err 0
fastrpc_apps_user.c:get_current_domain: tlsKey == INVALID_KEY
apps_std_imp.c:1307: Entering apps_std_get_search_paths_with_env
apps_std_imp.c:941: Entering get_dirlist_from_env
apps_std_imp.c:1016: Exiting get_dirlist_from_env for DSP_LIBRARY_PATH, err 0
apps_std_imp.c:1366: Exiting apps_std_get_search_paths_with_env for envvarname ADSP_LIBRARY_PATH delim ;, err 0
apps_std_imp.c:1415: Warniing 0x0: fileExists failed for path /home/val/src/fastrpc/test/v75/fastrpc_test.debugconfig, errno is No such file or directory

apps_std_imp.c:1415: Warniing 0x0: fileExists failed for path /usr/lib/dsp//fastrpc_test.debugconfig, errno is No such file or directory

fastrpc_config.c:418: fastrpc_config_init: Couldn't find file fastrpc_test.debugconfig, errno (No such file or directory) at /home/val/src/fastrpc/test/v75, /usr/lib/dsp/, 

platform_libs.c:23: calling init for mod_table
platform_libs.c:25: init for mod_table returned 0
platform_libs.c:23: calling init for apps_std
platform_libs.c:25: init for apps_std returned 0
fastrpc_apps_user.c:4258: fastrpc_apps_user_init done. default domain:3 and &fastrpc_trace:0xffff818de850

platform_libs.c:25: init for fastrpc_apps_user returned 0
fastrpc_apps_user.c:4379: multidsplib_env_init: libcdsprpc.so loaded
opening /home/val/src/fastrpc/test/linux/libcalculator.so: 0xaaaac1870c60

Allocate 4000 bytes from ION heap
rpcmem_linux.c:167: Allocted memory from DMA heap fd 5 ptr 0xffff819ac000 orig ptr 0xffff819ac000

Creating sequence of numbers from 0 to 999
Compute sum on domain 3
fastrpc_apps_user.c:2647: Entering remote_session_control, req 2, data 0xffffef31e658, size 8

fastrpc_apps_user.c:401: fastrpc_session_get, domain 3, state 0, ref 1

fastrpc_apps_user.c:2701: remote_session_control Unsigned PD enable 1 request for domain 3
fastrpc_apps_user.c:418: fastrpc_session_put, domain 3, state 0, ref 0

fastrpc_apps_user.c:2410: Entering remote_handle_control, req 2, data 0xffffef31e5d0, size 12

fastrpc_apps_user.c:401: fastrpc_session_get, domain 3, state 0, ref 1

fastrpc_apps_user.c:2291: Entering remote_handle_control_domain, domain 3, handle 18446744073709551615, req 2, data 0xffffef31e5d0, size 12

fastrpc_apps_user.c:418: fastrpc_session_put, domain 3, state 0, ref 0

fastrpc_apps_user.c:2647: Entering remote_session_control, req 11, data 0xffffef31e610, size 24

fastrpc_apps_user.c:401: fastrpc_session_get, domain 3, state 0, ref 1

fastrpc_apps_user.c:2856: remote_session_control Register PD status notification request for domain 3

fastrpc_apps_user.c:418: fastrpc_session_put, domain 3, state 0, ref 0

fastrpc_apps_user.c:1798: Entering remote_handle64_open, name file:///libcalculator_skel.so?calculator_skel_handle_invoke&_modver=1.0&_idlver=1.2.3&_dom=cdsp
[... 424 lines cut ...]
rpcmem_linux.c:167: Allocted memory from DMA heap fd 8 ptr 0xffff819aa000 orig ptr 0xffff819aa000

fastrpc_apps_user.c:3390: close_device_node: closed dev 5 on domain 3
apps_std_imp.c:452: Entering apps_std_fclose
apps_std_imp.c:243: Entering apps_std_FILE_get
apps_std_imp.c:259: Exiting apps_std_FILE_get fd 0x64 err 0
apps_std_imp.c:184: Entering apps_std_FILE_free
apps_std_imp.c:190: Exiting apps_std_FILE_free
apps_std_imp.c:472: Exiting apps_std_fclose sin 64 err 0
fastrpc_apps_user.c:3881: Error 0x8000054f: remote_init failed for domain 3, errno No error information, ioErr -2147482289

fastrpc_apps_user.c:3884: Done with remote_init, err: 0x8000054f, dev: -1
fastrpc_apps_user.c:3178: domain_deinit for domain 3: dev -1
fastrpc_apps_user.c:4094: Error 0x8000054f: domain_init (-1) failed for domain 3 (errno No error information)

fastrpc_apps_user.c:418: fastrpc_session_put, domain 3, state 0, ref 0

fastrpc_apps_user.c:1830: Error 0x8000054f: remote_handle64_open failed for file:///libhap_example_skel.so?hap_example_skel_handle_invoke&_modver=1.0&_dom=cdsp (errno No error information)

ERROR 0x8000054f: Unable to create FastRPC session on domain 3
```

</details>

(somehow only after recompiling, `get_current_domain` started crashing due to `tlsKey == INVALID_KEY` and `pthread_getspecific` being called on that o.0)

quic-ekangupt, 2025-07-02:
Hi Val,

The error code here(0x8000054f) suggests that PD creation on DSP is failing but the exact reason cannot be found from logs. Few of the possibilities include shell-firmware mismatch or some resource limitation from DSP. Can you tell me from where are you getting the cdsp.mbn(firmware) and fastrpc_shell* files?

I'm currently testing a change which might help in routing of the DSP logs. The patch will be posted by this week which might help in getting more details on this failure,

quic-ekangupt, 2025-07-10:
Hi Val,

I've created a new issue for updates on DSP headers.
https://github.com/qualcomm/fastrpc/issues/193

I'm closing this issue. Please let me know if any more information is needed.

Thanks,
Ekansh